bluerobotics/BlueRobotics_MS5837_Library #1: Library compiling error. constexpr does not name a type

cgirongu, 2016-06-27:
Hello,

I'm a basic user and wanted to try this library, but when I try compiling it along with my sketch it shows the error at the end of the post. 

I also get the error when I use the example so I'm sure the problem is in the library itself. 

Thank for sharing it and hope you find the bug as I could really take advantage of it! 

*****************************************************************************************************************************

In file included from MS5837_Example.ino:39:0:
C:\Users\cgiron\Documents\Arduino\libraries\BlueRobotics_MS5837_Library-master/MS5837.h:44:9: error: 'constexpr' does not name a type
  static constexpr float Pa = 100.0f;
         ^
C:\Users\cgiron\Documents\Arduino\libraries\BlueRobotics_MS5837_Library-master/MS5837.h:44:9: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
C:\Users\cgiron\Documents\Arduino\libraries\BlueRobotics_MS5837_Library-master/MS5837.h:45:9: error: 'constexpr' does not name a type
  static constexpr float bar = 0.001f;
         ^
C:\Users\cgiron\Documents\Arduino\libraries\BlueRobotics_MS5837_Library-master/MS5837.h:45:9: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
C:\Users\cgiron\Documents\Arduino\libraries\BlueRobotics_MS5837_Library-master/MS5837.h:46:9: error: 'constexpr' does not name a type
  static constexpr float mbar = 1.0f;
         ^
C:\Users\cgiron\Documents\Arduino\libraries\BlueRobotics_MS5837_Library-master/MS5837.h:46:9: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
Error compiling.

rjehangir, 2016-06-27:
Hi @cgirongu, I think this will be resolved if you update to a newer version of Arduino. The library does not work well for versions before 1.6. Please let me know if that the problem continues!

cgirongu, 2016-06-27:
Wrong button, 

I'm using arduino 1.7.10 

I'm trying to trace what the problem is in the library code but it seems to be very straight forward. I'll try using a 1.6 version of Arduino IDE and let you know

rjehangir, 2016-06-27:
@cgirongu The current release version of Arduino is 1.6.9, so I'm not sure what you are using. I'm not familiar with the dev versions of Arduino if that's what you are using.

cgirongu, 2016-06-27:
I was using a developer version and as soon as i moved back to 1.6.9 it worked perfectly! 

Thanks for the quick answer

closed 2016-06-27 by cgirongu