bluerobotics/BlueRobotics_MS5837_Library #2: CRC Calculation Function

crispc85, 2016-07-28:
Line 174 in MS5837.cpp is as follows:

`n_rem ^= (uint16_t)((n_prom[i>>1]) & 0x00FF0);`

However, this differs from the MS5837 manual which give the CRC function for that line as

`n_rem ^= (uint16_t)((n_prom[i>>1]) & 0x00FF);`

I don't know if this causes a problem on Arduino, but when I copied the code for use in a Linux Beaglebone project I was getting checksum errors until I noticed the extra zero in that hex value from the Github repository code. I know this is an Arduino library, but it seems to me that the CRC function won't provide the correct answer using an Arduino either without removing that extra `0` from the hex value on that line. Again, I might be wrong and it might work as expected on Arduino, just pointing out a discrepancy between the manual and the Github code.

Cheers!

Toffe87, 2017-03-10:
I did actually just port this to Node.JS and got the same error, i created a pull request for it. that extra zero.

It actually has something to say, it wont accept the checksum with it.

1 & 0x00FF = 1
1 & 0x00FF0 = 0

jaxxzer, 2017-04-10:
Thanks guys, https://github.com/bluerobotics/BlueRobotics_MS5837_Library/pull/4 fixed it.

closed 2017-04-10 by jaxxzer