bluerobotics/BlueRobotics_MS5837_Library #3: Only 7 prom registers

jaxxzer, 2017-01-18:
We have an array of size 8 to hold the internal prom, but the device only has 7 prom registers.

Toffe87, 2017-03-11:
It has eight, datasheet states from : 0xA0 to 0xAE, each 2 bytes. 

1: 0xA0 & 0xA1
2: 0xA2 & 0xA3
3: 0xA4 & 0xA5
4: 0xA6 & 0xA7
5: 0xA8 & 0xA9
6: 0xAA & 0xAB
7: 0xAC & 0xAD
8: 0xAE & 0xAF

I get an error reading AE to AF but i get two bytes anyhow. So there must be something working on the 8th byte set.

jaxxzer, 2017-04-10:
The word at 0xAE isn't defined anywhere in the datasheet except vaguely as a 'subsidiary value' in the crc example. Reading this value is a waste of resources as it is set to zero before proceeding with crc calculation. The code has been updated to reflect this.

closed 2017-04-10 by jaxxzer