bluerobotics/BlueRobotics_MS5837_Library #6: Not getting valid data.

RyanBone123, 2017-11-19:
Hi, 

first, I am not sure if this is the right place to post this. But I'll give it a try anyway.

I have been using this sensor with this library for quite some time and all of the sudden we started getting really weird data back from our device. I uploaded a picture of the serial data we are receiving from the example sketch.

Pressure: -20053014.00 mbar
Temperature: -389.20 deg C
Depth: -198730.95 m
Altitude: nan m above mean sea level

If we dunk it under water, heat it up or cool it down, this is the numbers we get. 

Any Idea of what went wrong? and way I can try and reset the device?

Thanks!
Ryan

rjehangir, 2017-11-19:
Hi Ryan, these look roughly like the numbers you'd get if the device is not responding at all. It sounds like it may be damaged in some way. Do you get the same results if the device is not plugged in?

RyanBone123, 2017-11-19:
So yes, that is in fact the data i get once I let it initialize, and then unplug it.

I guess it's a blown sensor.. likely my fault. 

Thanks Rusty!

closed 2017-12-11 by jaxxzer