bluerobotics/BlueRobotics_MS5837_Library #7: Mbar resolution

funkzy, 2018-05-03:
Hi,

Thanks for writing this awesome library!

One thing I noticed is that the sensor datasheet claims .2 mbar accuracy, but I'm only seeing whole number mbar numbers coming out of pressure(). Any thoughts on this?

Setup is pretty standard - using a Moteino (standard Arduino clone, ATMega328p, 3.3v) and running the unmodified sketch. Application is water depth sensing.

Anyways, it would be awesome if it's possible to get more resolution out of the sensor and get 2mm resolution.

Thanks!
Jeff

jaxxzer, 2018-05-03:
Thank you for your input. We will look to see if the output can be improved.

dwblair, 2018-10-03:
Hi All!

Thanks, also, BR, for this library!

@jaxxzer -- I also noticed the resolution issue.  I looked into the code and the associated MS5837 [data sheet](https://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Data+Sheet%7FMS5837-02BA01%7FA5%7Fpdf%7FEnglish%7FENG_DS_MS5837-02BA01_A5.pdf%7FCAT-BLPS0059) and I believe that the issue is caused by doing 'integer division' to get the final pressure value rather than 'floating point division'.

I've altered to library to return the full resolution of the sensor.  My version of the library is [here](https://github.com/edgecollective/BlueRobotics_MS5837_Library);  my pull request against the original is [here](https://github.com/bluerobotics/BlueRobotics_MS5837_Library/pull/9).

Cheers!

Don
[edgecollective](edgecollective.io)

closed 2019-03-19 by jaxxzer