bluerobotics/BlueRobotics_MS5837_Library #10: Add oversampling setting adjustment

jaxxzer, 2019-01-30:
This will allow us to reduce the blocking measurement [delay](https://github.com/bluerobotics/BlueRobotics_MS5837_Library/blob/master/MS5837.cpp#L66) time.

jaxxzer, 2019-01-30:
For example, our python library already supports this: https://github.com/bluerobotics/ms5837-python#readoversamplingosr_8192

senceryazici, 2019-05-13:
Absolutely, right now we're having issues using our bar30 sensor, as the sensor spends time of 40ms to read, it reduces our communication rate down to ~50Hz, 

since the oversampling can be set to 256, it can be running at 1.5Khz.
Please implement this feature to the arduino library.

Thanks

senceryazici, 2019-05-13:
I've modified such changes on my forked repository. => https://github.com/senceryazici/BlueRobotics_MS5837_Library/tree/pr_osr_implementation

I can open a pull request if you like.

Thanks

jaxxzer, 2019-05-13:
> I can open a pull request if you like.

@senceryazici yes please :+1:

senceryazici, 2019-05-14:
Pending pull request #13

kiteborne, 2020-04-10:
Why does it not compile on teensy 3.2 ? - is it some of the math in the library (POW) forexample ?