bluerobotics/BlueRobotics_MS5837_Library #19: Newest commit is not reflected in the arduino library manager

lmedagoda, 2020-10-16:
Hi, this took a day of my life trying to debug. Basically I was trying to get the full resolution of the Bar 02, but it wasn't working. Turns out that the newest commit (to get the full resolution of the Bar02) is not reflected when you automatically download from the arduino manger! I guess because the tag 1.1.0 refers to the Jun 2017 build? Anyways, download the library from here directly and put it in your Arduino library path so you can get the full resolution for the Bar 02. 

Perhaps other people are getting this issue and not noticing, since it is subtle and we only noticed it because we were using it as an altitude sensor. Underwater the reduced resolution would not be immediately apparent, its 16mm versus the true 0.16mm, since noise is still an issue to overcome.

Also, hopefully the developer might fix the tag in the arduino library to point to the updated version.

RedshiftVelocities, 2021-10-17:
A year later (huh, pretty much exactly, look at that!) this is still an issue. Blue Robotics, could you please update this library in the Arduino library manager? Thanks!

skotagiri, 2023-01-16:
Another request to release a version for Arduino Library manager with latest code.