bluerobotics/BlueRobotics_MS5837_Library #20: Library does not compile for certain newer Arduino targets

NeilBaner, 2020-12-31:
When I include the library in Arduino IDE, and compile for the Arduino Nano 33 BLE/BLE Sense boards, I get some errors due to `D1` and `D2` being #define-d in `pins_arduino.h`, from the built-in Arduino libraries, causing a conflict with the `D1` and `D2` variables declared in the library. 

The exact error is as follows:

```
In file included from C:\Users\neilbaner\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\cores\arduino/Arduino.h:90:0,
                 from sketch\MS5837_Example.ino.cpp:1:
C:\Users\neilbaner\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.3.0\variants\ARDUINO_NANO33BLE/pins_arduino.h:82:14: error: expected unqualified-id before numeric constant
 #define D1   1
              ^
C:\Users\neilbaner\Documents\Arduino\libraries\BlueRobotics_MS5837_Library/MS5837.h:88:11: note: in expansion of macro 'D1'
  uint32_t D1, D2;
           ^
exit status 1
Error compiling for board Arduino Nano 33 BLE.
```

Subseaking, 2022-12-12:
Same issue with the Portenta H7. Tried replacing all instances of D1 and D2 vars in the MS5837 lib, which I suppose fixed the var conflict but introduced a ton of new errors.
Any fix coming? Or does someone have a workaround to share? It would be much appreciated.

Subseaking, 2022-12-12:
Had the exact same issue on Portenta H7 - applied the changes suggested by @NeilBaner for merge and the code compiles now. Physical testing of sensor remains.