bluerobotics/BlueRobotics_MS5837_Library #29: Allow setting the ground level atmospheric pressure

ES-Alexander, 2023-03-17:
We currently offset by mean sea level pressure when estimating depth or altitude, but most applications get more useful data by using an explicit measured pressure value to offset by. Library users currently need to do that offsetting manually, which is inconvenient and somewhat unintuitive.

It would be preferable if we provided a suitable method for setting the current pressure as the basis to measure depth/altitude from, and simply use MSL as the _default_ offset value.

Issue mentioned in [this forum post](https://discuss.bluerobotics.com/t/bm30-mistake-sensor/13843), along with a variety of others previously (e.g. [this comment](https://discuss.bluerobotics.com/t/i2c-level-converter-with-bar30-pressure-sensor/797/43) from 2017).

This kind of offsetting / calibration is [already done in ArduPilot vehicle firmwares](https://github.com/ArduPilot/ardupilot/blob/Sub-4.1/libraries/AP_Baro/AP_Baro.cpp#L844-L853), and used regularly to set the zero-point for ArduSub vehicle depth, and the launch altitude for aerial vehicles.

allacmc, 2023-03-19:
Was this feature made?

Which function is used? To apply ?

ES-Alexander, 2023-03-22:
> Was this feature made?

No, not yet. This thread is a GitHub "issue", which provides information about a problem and/or a proposal/suggestion. The issue will get closed if/when the problem is solved / the feature gets implemented.

codymarquardt, 2026-11-11:
Is this still of interest?

ES-Alexander, 2026-11-11:
Sure - the idea is still useful, it just hasn't been implemented yet.

A PR would be welcome - likely adding something like a `setReferencePressure` method that sets an internal variable the `depth` and `altitude` methods then refer to (similar to how the `depth` method already refers to `fluidDensity`).