bluerobotics/BlueRobotics_MS5837_Library #30: Is altitude dependent on barometric pressure

BJPatriotRacing, 2023-05-19:
Thanks for this library, it's easy to implement and seems to work well. I've read the data sheet and I don't see any information if altitude is dependent on barometric pressure. 

Reason I ask is 

a) day 1 altitude is 700 feet above sea level--clear day, but I did not check the pressure
b) day 2 altitude was 505 feet above sea level (tested in the same spot), cloudy, near rain

I can only guess that the unit computes altitude based on barometric pressure.

Thoughts?

Kris

rafaellehmkuhl, 2023-05-23:
For accurate results underwater you should always calibrate your out-of-water pressure before diving.

This is how it's done in Ardupilot, for example.

ES-Alexander, 2023-12-04:
Just to close this out:
- yes, the altitude+depth estimates are calculated based on the sensor's measured pressure and temperature
- when using these sensors it's recommended to set the surface/ground pressure as the zero-point, like the 'tare' function on a scale, so that the relative depth/altitude estimates are correct for your operating conditions
    - this functionality is [not yet built into the library](#29)
- the 30BA sensor variant can have quite significant absolute error (up to ±200mbar) -> it's not expected to be a highly accurate absolute pressure sensor