bluerobotics/BlueRobotics_MS5837_Library #32: Bug: MS5837_30BA26 300 bar sensor detected as 2 bar sensor

scuba-hacker, 2024-01-12:
I switched from using the current Arduino library version 1.1.1 of MS58357.h/.cpp to head of master (I forked the project) and found that depth was being reported as zero despite being several metres under water. This is for the 300 bar sensor, bought just under a year ago.

It turns out that the model detection code in the init() function is assigning model = MS5837_02BA (value 1) instead of = MS5837_30BA (value 0). I am not calling setModel() anywhere in the client code. It seems that something in the "Read calibration values and CRC" is not working.

The current Arduino library version doesn't have this model check and _model defaults to MS5837_30BA. As a separate issue none of the private member variables are being initialised in the constructor so their values are officially undefined (and fortunately the compiler is setting them to zero automatically).  They should be initialised in the constructor to 0, including the C[8] array contents.

ES-Alexander, 2025-05-14:
I happened to come across this, and the auto-detection of the model type that apparently [got added 4 years ago](https://github.com/bluerobotics/BlueRobotics_MS5837_Library/commit/65ebe94589f1a42f3020592f99ac1fc7f18bae4b) has recently been [determined to be invalid](https://github.com/ArduPilot/ardupilot/pull/29122#issuecomment-2648486046). There's some work going on in the ArduPilot firmware to provide a heuristic-based approach instead (details in that linked thread), which we should be able to transfer here (and to the Python library) once it's confirmed and merged.

scuba-hacker, 2025-05-16:
Sounds good!

closed 2025-06-30 by ES-Alexander